guyc/py-gaugette #12: Platform detection fails for Raspberry Pi 2

MartyMacGyver, 2015-02-17:
Platform detection currently depends on whether platform.machine() returns 'armv7l'. This fails on the Pi2 (which, like the BBB will detect as 'armv7l'). An example of a working and more comprehensive solution to the platform detection problem is found in the current Adafruit_Python_GPIO platform detection code - this also handles Pi revision detection which may be useful down the road.

https://github.com/adafruit/Adafruit_Python_GPIO/blob/master/Adafruit_GPIO/Platform.py

guyc, 2015-03-14:
Thanks for drawing that to my attention.
So Adafruit have a multi-platform GPIO implementation?  Cool, I didn't know that.  I should probably swap out my entire GPIO abstraction layer for theirs.  For now I'll just get my platform detection fixed using their logic.

guyc, 2015-03-14:
Resolved with commit c20cf020c9ccad8c453f2de4460d1a24c383e297

closed 2015-03-14 by guyc