guyc/py-gaugette #15: Random pixel / previous image

JokerQyou, 2015-12-12:
First of all thank you for this awesome library. But I'm having trouble using it.
I purchased a SSD1306-based OLED screen which is `0.96 inch` with a resolution of `128x64`, and the wiring is as follows:

```plain
Screen pin                RaspberryPi (B+) pin
GND                         GND
VCC                         3V3
D0                          SCLK (#11)
D1                          MOSI (#10)
RST                         GPIO 1 (#18)
DC                          GPIO 0 (#17)
CS                          CE0 (#8)
```

I've fixed the platform detection problems in the sample codes, but if I run the `ssd1306_test.py` I only get previous displayed image scrolling (speaking of that, the scrolling is having its own problem). Now I'll explain 'previous image': when I first bought this screen, I've tested with two other libraries to see if the hardware is ok, these are [Adafruit SSD1306 python library][1] and [AdaPi library][2] (from which I found this library by you). Both library works fine, the shapes / text / line / image showed up correctly on the OLED screen. The only problem being that these example codes did not cleanup the screen after they exit, thus leaving the screen with an image. When I test with this library, I get nothing but the image left by other libraries. 

Also I can't find an efficient way to get rid of these images, except for entirely disconnect all pins from my RaspberryPi B+. By doing so, I can get a clean screen with no pixels lighted up. If I run the sample code in this library now, the only thing I get is some random pixels.

I hope you could help me, or just give some hints about what might be wrong.

  [1]: https://github.com/adafruit/Adafruit_Python_SSD1306
  [2]: https://github.com/andig/AdaPi

JokerQyou, 2015-12-12:
I also found this comment on your blog, which could be the same issue as this one, http://guy.carpenter.id.au/gaugette/2012/11/08/controlling-an-adafruit-spi-oled-with-a-raspberry-pi/#comment-1045718087

JokerQyou, 2015-12-12:
More information: If I go with the `font_test.py` in sample code directory, the screen will light up with random pixles, but after one scroll or more, it will go completly dark, as if all the lighted points are outside the screen.

guyc, 2015-12-12:
Thanks for cleaning up the library, it's very much appreciated.  Seems I was not very careful when I dropped in the new platform detection library.  

Your problem sounds like comms just aren't working.  I'll try to find time to replicate your setup tomorrow and see if I can reproduce the problem.

guyc, 2015-12-13:
I'm afraid I can't reproduce the problem.  I fired up a B+ with Raspbian 3.18.11-v7+ and did a clean install of the dependencies and sources, and enabled SPI in /boot/config.txt.   

I used the following wiring which requires no changes to the pin assignments in the committed sample code:

Module Pin | Module Signal  |  RPi Pin | RPi Signal
-------------- | ------------- | ------------- | ------------
1 | GND | 6 | GND
2 | VIN   |  1  |  3.3V
3 | 3.3v   |     | NC
4 | CS     | 24 |  GPIO8 SPI0_CE0_N
5 | RST   | 8   | GPIO14 UART0_TXD
6 | DC     | 10 |  GPIO15 UART0_RXD
7 | CLK   | 23 |  GPIO11 SPI0_SCLK
8 | DATA | 19  | GPIO10 SPI0_MOSI

The RPi pin numbers above are the mechanical pin numbers on the 40-pin GPIO header.  Looks like these differ from your pin connections in the choices of RST and DC, and it seems your module has different screen-print labels for DATA and CLK.

I used a 128x64 module from Adafruit.  ssd1306_test.py by default assumes a 32-row display, which renders alternate pixel rows (but is legible and scrolls fine).  Setting ROWS to 64 fixes that.

I don't know what to suggest other than to double-check your wiring.  

With regard to clearing the screen, to my knowledge the only way to clear the screen on these devices is to clock in 0 values for all pixels.

JokerQyou, 2015-12-13:
O thank you so much for the great help! That definitely saved my weekend! After carefully reading your wiring, it turns out that I had misunderstood the pin definition in your library. I have wiringPi installed, and the result of `sudo gpio readall` is as follows:

```plain
 +-----+-----+---------+------+---+--B Plus--+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 |     |     |    3.3v |      |   |  1 || 2  |   |      | 5v      |     |     |
 |   2 |   8 |   SDA.1 | ALT0 | 1 |  3 || 4  |   |      | 5V      |     |     |
 |   3 |   9 |   SCL.1 | ALT0 | 1 |  5 || 6  |   |      | 0v      |     |     |
 |   4 |   7 | GPIO. 7 |   IN | 1 |  7 || 8  | 1 | ALT0 | TxD     | 15  | 14  |
 |     |     |      0v |      |   |  9 || 10 | 1 | ALT0 | RxD     | 16  | 15  |
 |  17 |   0 | GPIO. 0 |   IN | 0 | 11 || 12 | 0 | IN   | GPIO. 1 | 1   | 18  |
 |  27 |   2 | GPIO. 2 |   IN | 0 | 13 || 14 |   |      | 0v      |     |     |
 |  22 |   3 | GPIO. 3 |   IN | 0 | 15 || 16 | 0 | IN   | GPIO. 4 | 4   | 23  |
 |     |     |    3.3v |      |   | 17 || 18 | 0 | IN   | GPIO. 5 | 5   | 24  |
 |  10 |  12 |    MOSI | ALT0 | 1 | 19 || 20 |   |      | 0v      |     |     |
 |   9 |  13 |    MISO | ALT0 | 0 | 21 || 22 | 0 | IN   | GPIO. 6 | 6   | 25  |
 |  11 |  14 |    SCLK | ALT0 | 0 | 23 || 24 | 1 | ALT0 | CE0     | 10  | 8   |
 |     |     |      0v |      |   | 25 || 26 | 1 | ALT0 | CE1     | 11  | 7   |
 |   0 |  30 |   SDA.0 |   IN | 1 | 27 || 28 | 1 | IN   | SCL.0   | 31  | 1   |
 |   5 |  21 | GPIO.21 |   IN | 1 | 29 || 30 |   |      | 0v      |     |     |
 |   6 |  22 | GPIO.22 |   IN | 1 | 31 || 32 | 0 | IN   | GPIO.26 | 26  | 12  |
 |  13 |  23 | GPIO.23 |   IN | 0 | 33 || 34 |   |      | 0v      |     |     |
 |  19 |  24 | GPIO.24 |   IN | 0 | 35 || 36 | 0 | IN   | GPIO.27 | 27  | 16  |
 |  26 |  25 | GPIO.25 |   IN | 0 | 37 || 38 | 0 | IN   | GPIO.28 | 28  | 20  |
 |     |     |      0v |      |   | 39 || 40 | 0 | IN   | GPIO.29 | 29  | 21  |
 +-----+-----+---------+------+---+----++----+---+------+---------+-----+-----+
 | BCM | wPi |   Name  | Mode | V | Physical | V | Mode | Name    | wPi | BCM |
 +-----+-----+---------+------+---+--B Plus--+---+------+---------+-----+-----+
```

And I have had RST connected to GPIO 1 and DC connected to GPIO 0, but I filled BCM pin numbers into the test script (RST = 18, DC = 17), which should be wPi pin numbers instead (RST = 1, DC = 0).

closed 2015-12-13 by JokerQyou