guyc/py-gaugette #24: Platform detection fails on RPi with arch linux arm

Keridos, 2017-03-28:
I get this error: 
Traceback (most recent call last):
  File "rotenctest.py", line 4, in <module>
    import gaugette.gpio
  File "/usr/lib/python3.6/site-packages/gaugette/gpio.py", line 12, in <module>
    import gaugette.platform
  File "/usr/lib/python3.6/site-packages/gaugette/platform.py", line 100, in <module>
    platform = platform_detect()
  File "/usr/lib/python3.6/site-packages/gaugette/platform.py", line 42, in platform_detect
    plat = platform.platform()
NameError: name 'platform' is not defined

with the default code for the program you have posted for a rotary encoder.

guyc, 2017-03-28:
Is your hardware RPi3?

Can you post the results of `cat /proc/cpuinfo` please?

guyc, 2017-03-28:
If it is a Pi3, PR #25 might solve the problem.

Keridos, 2017-03-31:
No its a raspberry pi model B, the old one.

guyc, 2017-04-03:
@Keridos are you able to provide the output from `cat /proc/cpuinfo`?

guyc, 2017-08-25:
fixed in commit df9677863c9512b4d8c9ae32838d90973573efb9

closed 2017-08-25 by guyc