guyc/py-gaugette #30: i2cdetect showing all addresses filled because of py-gaugette

kramer65, 2017-06-09:
I'm using py-gaugette to read out a rotary encoder from my pi. It works fine, but after the last of these five lines of code:
```
>>> import gaugette.gpio
>>> import gaugette.rotary_encoder
>>> import gaugette
>>> gpio = gaugette.gpio.GPIO()
>>> encoder = gaugette.rotary_encoder.RotaryEncoder(gpio, 7, 8)
```
suddenly `i2cdetect` shows every possible address:
```
pi@raspberrypi:~$ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          03 04 05 06 07 08 09 0a 0b 0c 0d 0e 0f
10: 10 11 12 13 14 15 16 17 18 19 1a 1b 1c 1d 1e 1f
20: 20 21 22 23 24 25 26 27 28 29 2a 2b 2c 2d 2e 2f
30: 30 31 32 33 34 35 36 37 38 39 3a 3b 3c 3d 3e 3f
40: 40 41 42 43 44 45 46 47 48 49 4a 4b 4c 4d 4e 4f
50: 50 51 52 53 54 55 56 57 58 59 5a 5b 5c 5d 5e 5f
60: 60 61 62 63 64 65 66 67 68 69 6a 6b 6c 6d 6e 6f
70: 70 71 72 73 74 75 76 77
```
The code after that (which reads out the rotary encoder) still works. Another connected thing (an IMU) doesn't work anymore though, although I don't know if this is the cause of that.

This behaviour also occurs if I don't connect anything to the raspi (leaving all GPIO pins unconnected). Even then all possible addresses seem filled after running this py-gaugette code.

Would you know why py-gaugette displays this behaviour of showing all addresses ?

guyc, 2017-06-23:
I haven't seen that.  I'll try to reproduce it when I get a chance to.

guyc, 2017-08-25:
I have a hunch I know what it is.  

https://raspberrypi.stackexchange.com/a/32720/7589
> Generally this happens because something is pulling SDA (pin 3) low.
> The i2cdetect program checks for the existence of a device by sending its address then letting SDA float high. If the device exists it should pull SDA low to signal its presence.

You are using wiring-pi pins 7 and 8.  Pin 8 is header pin 3 => SDA for I2C.

[image]

kramer65, 2017-08-29:
You're right!

I simply connected the rotary encoder to some other pins, and now it works like a charm.

Thanks!

closed 2017-08-29 by kramer65