guyc/py-gaugette #31: Instructions on README don't work

Psychokiller1888, 2017-06-22:
Hi!

The instructions don't work:

pi@raspberrypi:~/python/minidisplay $ sudo python display.py
Traceback (most recent call last):
  File "display.py", line 6, in <module>
    spi = spi = gaugette.spi.SPI(spi_bus=0, spi_device=0)
TypeError: __init__() got an unexpected keyword argument 'spi_device'

Correcting to correct declaration (gaugette.spi.SPI(bus=0, device=0)  leads you to:

pi@raspberrypi:~/python/minidisplay $ sudo python display.py
Traceback (most recent call last):
  File "display.py", line 6, in <module>
    spi = spi = gaugette.spi.SPI(bus=0, device=0)
  File "/usr/local/lib/python2.7/dist-packages/gaugette/spi.py", line 22, in __init__
    self.spi.open(bus, device)
IOError: [Errno 2] No such file or directory

Is this project still supported or deprected?

guyc, 2017-06-23:
Thanks for pointing out the required changes in the readme.

The error you are seeing suggests to me that `/dev/spidev0.0` doesn't exist.  Depending on your choice of OS and version, there is some configuration to required to enable it.  e.g. for Raspian: https://www.raspberrypi.org/documentation/hardware/raspberrypi/spi/README.md

Psychokiller1888, 2017-06-23:
Hmmm, dunno what I can say, I replugged my raspi today and it worked (after changing the call of course) I could swear I did reboot my raspberry after installing spidev, but who knows, I might be mistaken. I see you also took the double import away, good job! Thanks for this btw

Psychokiller1888, 2017-06-23:
Do you have any idea why I end up with this instead of the Hello World it should print according to the readme?

[image]

guyc, 2017-08-24:
I'm going to close this as the original issue has been resolved.

closed 2017-08-24 by guyc