guyc/py-gaugette #33: Python3 Segmentation Fault with Rotary Encoder

OliverMardle, 2017-08-23:
Hi, I just tried to run the example program for the rotary encoder provided and as soon as I turn the rotary encoder the program terminates and gives the "Segmentation Fault" error.

Cheers.

guyc, 2017-08-23:
There are three samples for the rotary encoder.  Which one gives you an error?

1. https://github.com/guyc/py-gaugette/blob/master/samples/rotary_test.py
2. https://github.com/guyc/py-gaugette/blob/master/samples/rotary_worker_test.py
3. https://github.com/guyc/py-gaugette/blob/master/samples/rotary_interrupt_test.py
 
My money is on (3).  Do the others give the same result?

It would be good to know some more details.
 - what board you are using
 - what OS you are using
 - output from `python --version`
 - output from `pip show wiringpi`

OliverMardle, 2017-08-24:
Hi, Sorry for the complete lack of information I provided... Your absolutely right, the issue occurs on rotary_interrupt_test.py. The switch works but as soon as I turn the dial I get the error.
I'm using a Raspberry Pi 2 at the moment and the latest distribution of Raspbian Stretch. In terms of python I'm using Python 3 and the output of `python3 --version` is `Python 3.5.3`
The output of `sudo pip3 show wiringpi` is:
```
Name: wiringpi
Version: 2.44.4
Summary: UNKNOWN
Home-page: UNKNOWN
Author: UNKNOWN
Author-email: UNKNOWN
License: UNKNOWN
Location: /usr/local/lib/python3.5/dist-packages
Requires:
```
When I do run the other working programs the rotary encoder appears to be a bit jumpy, I think this must be the stage of the encoder however this can be resolved at a later stage.

Cheers.