guyc/py-gaugette #36: SSD1306 OLED didn't work

jaschkopf, 2018-05-30:
Hello. I have installes your lib with [this tutorial](https://github.com/the-raspberry-pi-guy/OLED) from "the raspberry pi guy". I'm using a brand new Rapsberry Pi Zero W and a Adafruit 326 128x64 OLED Display with freshly installes Raspbian Stretch Lite 4.14. Maybe I'm to stupid to make it run, but wenn I run the examples from the raspberry pi guy, the screen stays blank. To check my hardware I installed the [Adafruit Python SSD1306](https://github.com/adafruit/Adafruit_Python_SSD1306) lib and the examples run sucessfully.

Now I'm wondering, if there is a problem with my way to install your lib or maybe a bug in any of the packages. 

I really like your lib and I'm building a project which is using py-gaugette, so I really need to run it on my Raspi Zero to finish it. 

Sorry for my bad english -.-
Thank you very much :)

guyc, 2018-05-30:
Okay, not much to go on, but here's where I would start:

Are you using i2c or SPI?  It looks like that display supports both, but by default that display uses SPI.  Hopefully you are using SPI.

The most likely problem is the pin-out.  
The py guy demo configures pins like this:
```
RESET_PIN = 15 # WiringPi pin 15 is GPIO14.
DC_PIN = 16 # WiringPi pin 16 is GPIO15.
```
here: https://github.com/the-raspberry-pi-guy/OLED/blob/master/python-examples/OLEDclock.py#L17
This is correct for the wiring here:
https://guy.carpenter.id.au/gaugette/2012/11/08/controlling-an-adafruit-spi-oled-with-a-raspberry-pi/

Adafruit configures pins like this:
```
RST = 24
DC = 23
```
here: https://github.com/adafruit/Adafruit_Python_SSD1306/blob/master/examples/animate.py#L33
For the wiring here:
https://learn.adafruit.com/ssd1306-oled-displays-with-raspberry-pi-and-beaglebone-black/wiring

Looks like you need to change the pin numbers in the pi-guy's demos.  The pin numbers are tricky because there are several different numbering sequences, and Adafruit does not use the same sequence as py-gaugette, which uses wiringPi numbering, see https://projects.drogon.net/raspberry-pi/wiringpi/pins/

At a glance I think you want:
RESET_PIN = 5
DC_PIN = 4

jaschkopf, 2018-05-30:
I have tried it, but nothing changed. The display is not wired like the Adafruit tutorial. RST = GPIO 24 and DC = GPIO23. I changed the OLEDclock.py like this:
`# OLEDclock.py

# This program interfaces with one of Adafruit's OLED displays and a Raspberry Pi (over SPI). It displays the current 
# date (Day, Month, Year) and then scrolls to the current time. The program waits for 2 seconds between scrolls.

# Example code from the py-gaugette library... Commented by The Raspberry Pi Guy

# Imports the necessary modules
import gaugette.ssd1306
import gaugette.platform
import gaugette.gpio
import time
import sys

# Define which GPIO pins the reset (RST) and DC signals on the OLED display are connected to on the
# Raspberry Pi. The defined pin numbers must use the WiringPi pin numbering scheme.
RESET_PIN = 5 # WiringPi pin 15 is GPIO14.
DC_PIN = 4 # WiringPi pin 16 is GPIO15.

gpio = gaugette.gpio.GPIO()
spi = gaugette.spi.SPI(bus=0, device=0)

# Very important... This lets py-gaugette 'know' what pins to use in order to reset the display
led = gaugette.ssd1306.SSD1306(gpio, spi, reset_pin=RESET_PIN, dc_pin=DC_PIN, rows=64, cols=128) # Change rows & cols values depending on your display dimensions.
led.begin()
led.clear_display()
led.display()
led.invert_display()
time.sleep(0.5)
led.normal_display()
time.sleep(0.5)

offset = 32 # flips between 0 and 32 for double buffering

# While loop has bulk of the code in it!

while True:

    # write the current time to the display on every other cycle
    if offset == 0:
        text = time.strftime("%A")
        led.draw_text2(0,0,text,2)
        text = time.strftime("%e %b %Y")
        led.draw_text2(0,16,text,2)
        text = time.strftime("%X")
        led.draw_text2(0,32+4,text,3)
        led.display()
        time.sleep(1)
    else:
        time.sleep(1)

    # vertically scroll to switch between buffers
    for i in range(0,32):
        offset = (offset + 1) % 64
        led.command(led.SET_START_LINE | offset)
        time.sleep(0.01)
`
Any idea?

jaschkopf, 2018-05-30:
I tried both. It is just build up on a bread board. Now it is wired like this:

DISPLAY-RASPI

GND-GND
VIN-3.3V
CS-CE0 (GPIO 8)
RST-GPIO 24
DC-GPIO23
DATA-MOSI (GPIO 10)
CLK-CLK (GPIO11)

Like this it works with the Adafruit lib, but not with py-gaugette

guyc, 2018-05-30:
Well now I'm stumped.  I haven't run pi-guy's demos.  You might try the demo from py-gaugette:
https://github.com/guyc/py-gaugette/blob/master/samples/ssd1306_test.py
but of course you must change the pin selections.

To be honest I doubt that will make a difference - pi-guy's code looks virtually identical.  I suppose it is possible that the most recent versions of this display have changed specs.  Seems unlikely, but not sure what else to suggest.

jaschkopf, 2018-05-31:
Soooo I have some good news. I found out, that the screen I used is a SSD1309. I tried an other OLED screen with a SSD1306 IC and it works! 

BUT

For my project I would like to use the SSD1309, because it's bigger (2.42"). Is there any chance to adapt your project to a SSD1309 display? 

Maybe it's just a little change, because with Adafuit and lumacore the SSD1309 works even with the SSD1306 setup, no changes needed.  

Thanks for your help so far 👍

guyc, 2018-05-31:
Ahhh.  That's interesting.  Glad you found it.  I haven't found anything online discussing the difference between the two controllers or gives any hits at what would need to change.  If you can find anything that explains the difference, let me know.

jaschkopf, 2018-06-01:
Hey. I found some datasheets of the two drivers. Maybe you cold found any significant diferences between them. I compared them and they are mostly the same, only a feaw commands are different. Because the SSD1309 didn't even turn on, I think it must be something wrong with the power on command. 

[SSD1306_v1.1.pdf](https://github.com/guyc/py-gaugette/files/2062305/SSD1306_v1.1.pdf)
[SSD1309_v1.1.pdf](https://github.com/guyc/py-gaugette/files/2062306/SSD1309_v1.1.pdf)

I really hope you can hepl me. Thank you very much

closed 2018-07-15 by guyc