guyc/py-gaugette #39: jetson nano (NVIDIA) ARM Cortex-A57, Platform is not supported.

Avi-avidan, 2020-11-08:
hi,
loved this repo.
I am trying to read dc motor encoder link, using your example, how ever I am getting NotImplementedError: Platform is not supported.

any advice will be appreciated.

this is the motor I am using -
https://www.aliexpress.com/i/4000098341909.html

this is jetson nano board -
https://developer.nvidia.com/embedded/jetson-nano-developer-kit

guyc, 2020-11-08:
The problem is that the gpio wrapper needs a new block to adapt the Jetson.GPIO library.
At a glance the Jetson library looks like a drop-in for Adafruit_BBIO.GPIO, so try changing the block for beagleboneblack
to support your platform:
https://github.com/guyc/py-gaugette/blob/14732a78fc953cfc920f548b04d69d5b6219ba54/gaugette/gpio.py#L34

eg experimentally change this:
```
elif gaugette.platform.isBeagleBoneBlack:
            import Adafruit_BBIO.GPIO
            self.gpio = Adafruit_BBIO.GPIO
```
to
```
else:
            import Jetson.GPIO
            self.gpio = Jetson.GPIO
```

Avi-avidan, 2020-11-09:
thanks :) that helped alot. moving down the rubbit hole reveals Jetson.GPIO has no attribute 'trigger'...

jpalves, 2024-04-25:
I have a modified version to work with Jetson Nano,  take a look https://github.com/jpalves/py-gaugette